MegaPirateNG/ardupilot-mpng #193: Crius GPS UBlox NEO 6M v3.1 - fly away

ther0x, 2015-01-21:
Dear Friends

I have bought a CRIUS (v3.1) UBlox NEO 6m GPS but I have a lot of problems
My quad, often, fly away when I use GPS modes (loiter and so on..)

Is there a right configuration for it? I have leaved it with original setup...

PS I use MegaPirateNG 3.1.5r2 on a crius v2, tried also with 3.0.1

Thanks!

N.B. I forgot, number of satellites (11) and hdop (< 2) are ok...

StenUhlig, 2015-01-21:
Hi,

I have the same setup (CRIUS AIO V2 and NEO 6) with MPNG 3.1.5 R2 and the same problem!! 

see https://github.com/MegaPirateNG/ardupilot-mpng/issues/194

Raqtor, 2015-02-26:
Make sure you have a SOLID GPS lock before arming your quad. Sometimes it can take a while so just let it stand there unarmed till you are sure it has and holds GPS fix.