MegaPirateNG/ardupilot-mpng #194: Loiter Fly Away Problem with MPNG 3.1.5 R2

StenUhlig, 2015-01-21:
Hi,

I run a CRIUS AIO V2 and NEO 6 with MPNG 3.1.5 R2.

Vibrations are Ok (I use an anti vibration mount). I run the GPS on UBLOX. 
I did  AutoTrim and AutoTuning successfully. 

The copter (quad) flies good in stabilize and in alt_hold!

As soon as I switch to Loiter, the copter flies away in random(?) directions.

Analyzing my logs did not help so far. 
hdop is below 2. 8-11 sats.

AutoAnalyzer says:
Test: Compass = GOOD - mag_field interference within limits (20.43%)
Test: GPS = GOOD - 

and so on.

What is wrong? Any idea? 

Thanks!!

ther0x, 2015-01-21:
My friend give me a second GPS (the same model) and I have to make some test
I left the original GPS setup

Let you know

ther0x, 2015-02-03:
Do you have made autotrim?

ther0x, 2015-02-05:
sorry but how do you have flushed your FC? from mission planner?

StenUhlig, 2015-02-05:
no, i did an EEPROMClear using arduino (http://arduino.cc/en/Tutorial/EEPROMClear) and reinstalled MPNG with the flash tool!

ar99, 2015-02-10:
EEPROM clear is the same that cli command setup -> erase?

StenUhlig, 2015-02-13:
I don't know!

SirAlex, 2015-02-13:
Yes, it's same operation.