MegaPirateNG/ardupilot-mpng #196: 3.1.5R2 last ?  Is there new MPNG version coming..Please comment

Espenf, 2015-03-10:
Will there be any progress in the code regarding the non-32bit boards?

smurfy, 2015-03-11:
@SirAlex stated a while back, there will be no new updates for 8bit.

The reason was and still is, that it's becomes harder and harder to adapt the code.
The 8bit processor is on it's limit as it is. the cheap all in one boards also use i2c instead of SPI which uses additional processor/io power which just is not available anymore.

You already can see that the official ardupilot versions leave features out for the 8bit versions because of this. (for example in arduplane some new algorythms)

But the project is an open source one, maybe someone has the time to try adapt the new code to work and somehow overcome the processor limits by removing some features, which will probably take a lot of time.

In my opinion the 8bit boards will start dying out. The 32bit boards gets cheaper and cheaper. (they are almost at the price i initially payed for my 8bit board 2 years ago)

SirAlex, 2015-03-11:
Hi Espenf, currently i have no plan to release new versions. If someone want to continue supporting 8bit - you are welcome.
As Smurfy mentioned, main problem is CPU load. If someone will cut unused features, maybe it will give us some power. For example, cut PWM inputs and leave only CPPM (it works faster). Remove serial ports and leave only Console/Mavlink on S0. And etc...

Espenf, 2015-03-12:
Ok, I understand. Tnx a lot for all your support during the last years and for still answering now and then! What's your current project / board for 32bit?

SirAlex, 2015-03-13:
Actually, I'm suspended my hobby now (Working on some hackwork) and I just answering questions and support build server. 
But F4BY team still working on it.

goial, 2015-06-06:
Copter 3.3-rc1 11-Apr-2015
Only support fast CPUs boards (Pixhawk, VRBrain, etc) and drop support for APM1, APM2 (sorry!)

When the official Arducopter stops for slow cpus at 3.2, i think it is okay, that MPNG 3.1.5 is the last release , because we have autotune!!! :D
Thank you for the great work on MPNG! :)