MegaPirateNG/ardupilot-mpng #198: No flashing with flashTool possible with Mac OS X

SirVival71, 2015-04-07:
Hello,
if I try to flash with the flashTool at a MacBook with OS X, I get the error:
>/Applications/Spielzeug/QuadroCopter/flashTool.app/Contents/MacOS/external/avrdude.exe -C/Applications/Spielzeug/QuadroCopter/flashTool.app/Contents/MacOS/external/avrdude.conf -patmega2560 -cwiring -Pusbserial-A8005M4z -b115200 -D -Uflash:w:/Applications/Spielzeug/QuadroCopter/flashTool.app/Contents/MacOS/firmwares/bf2a62f32703769dced35ca5e0157726_c469f7b9b3d9bc22a4fd57f50526d24d505e2124.hex:iavrdude.exe: ser_open(): can't open device "usbserial-A8005M4z": No such file or directory
ioctl("TIOCMGET"): Inappropriate ioctl for device
ioctl("TIOCMGET"): Inappropriate ioctl for device
avrdude.exe: ser_send(): write error: Bad file descriptor

Why does it not flash? A connect with a terminal (screen /dev/tty.usbserial-A8005M4z 115200) is possible, APM Planner 2 does work as well.
Is the problem, that the port is /dev/tty.usbserial-A8005M4z and not like in the log without tty.?
What can I do?

smurfy, 2015-04-07:
i'm not 100% sure but as soon i have access to my mac again i will try to reproduce the error.

SirVival71, 2015-05-01:
Can someone help me? Where does it search for the device?
I have /dev/cu.usbserial-A8005M4z and /dev/tty.usbserial-A8005M4z.
The port declaration in the flashtool is -Pusbserial-A8005M4z. Is this the problem? How can I change the port-declaration?

tezmo, 2015-05-10:
same issue here..