MegaPirateNG/ardupilot-mpng #200: megapirateNG.com down?

goial, 2015-05-15:
I cant use the flashtool and the Homepage is since 3 days unreachable :( Whats wrong guys? :'(
Lars

i--storm, 2015-05-15:
Same problem

smurfy, 2015-05-15:
sir alex mentioned something about moving servers or something like that.

schuppeste, 2015-05-21:
its possible to install own Flashtoolserver and forward port to local machine to compile Firmware for Flashtool?

smurfy, 2015-05-21:
In theory yes. Everything you need is in the FlashTool repository. Only thing is, you need to setup the compile environment and also recompile FlashTool to add the "new" server url.

smurfy, 2015-05-21:
Info: https://github.com/MegaPirateNG/FlashTool
I'm also happy to host the build-server

SirAlex, 2015-05-21:
Please, wait a moment. Server relocated to new place, give me time to adjust DNS settings and check all configurations.

SirAlex, 2015-05-21:
Server is up. But populating new DNS setting may take some time.

schuppeste, 2015-05-21:
The flashtool needs a inifile or commandline arguments to set alternative Servers.. To change simple Servers to other repos

smurfy, 2015-05-21:
yepp, thought the same. i have some more improvement ideas, but just no time yet.

danil2009, 2016-08-12:
Hi sir Alex
I want to back my firmware to megapirateNG. My quadcopter using Crius V2.0, now i ca not use flash tool. when you will be fix it or relocated the server Sir?