MegaPirateNG/ardupilot-mpng #203: Error while compiling 3.0.1R4

asaadcivic99, 2015-06-07:
Im a newbie and I have an HK AIO board. Im using Arduino version from the link in the megapirate site. and I get the following error while compiling the code:

In file included from D:\FPV\Quad\ArduPilot-Arduino-1.0.3-gcc-4.8.2-windows\ArduPilot-Arduino-1.0.3-windows\libraries\AP_HAL_PX4/AP_HAL_PX4.h:6:0,
                 from ArduCopter.pde:74:
D:\FPV\Quad\ArduPilot-Arduino-1.0.3-gcc-4.8.2-windows\ArduPilot-Arduino-1.0.3-windows\libraries\AP_HAL_PX4/HAL_PX4_Class.h:11:34: fatal error: systemlib/visibility.h: No such file or directory
 #include <systemlib/visibility.h>
                                  ^
compilation terminated.

CrashDBN, 2015-07-25:
Hi there. Why not use the flash tool to upload, way way easier than using Arduino.

Bazmundi, 2017-11-19:
@CrashDBN good job!  He hasn't compiled it yet so it won't matter what tool you use to flash.

@asaadcivic99 I am in same boat.  Downloaded tools and source, followed instructions.  Built fails at another sport to yours.