MegaPirateNG/ardupilot-mpng #205: HK AIOP board not start with ArduRover code

Metori, 2016-06-23:
I builded APMrover2 code on Ubuntu machine. Upload it with avrdude into my HK aiop board (Crius v1.1 clone). But seems that board can't start.

Red led flashes a bit, "Init ArduRo" string prints to UART and then nothing...

My board has been checked with MultiWii and MPNG (copter code). So hardware is working.

Espenf, 2016-06-23:
I have only used this board with MultiWi or MPNG, dont think APM will run. 
http://ardupilot.org/dev/docs/supported-autopilot-controller-boards.html

Metori, 2016-06-23:
There is a APMrover2 code in MPNG branch. I think it should work, because of it not removed from this branch.
In Issue https://github.com/MegaPirateNG/ardupilot-mpng/issues/132 smurfy sad that he launched MPNG ArduRover.