MegaPirateNG/ardupilot-mpng #208: Is Arduino Mega 2560 controller supported ?

Osama-321, 2019-09-09:
Can i use Arduino Mega2560 board to uplaod the megapirate code and  run it with mission planner ?

smurfy, 2019-09-10:
The chip itself is supported, but you need a lot of sensors to be really usable

Osama-321, 2019-09-10:
I tried to use it with only mpu6050 ... the megapirate code was uploaded successfully on it ... but when I connected it to mission planner ,an error message appeared (related to Mavlink ) 
so .. is the problem because I am did not connect the required sensors (like : hmc compass , barometer )
note : when I upload the code to my arduino I used the flash tool , and I chose the board type (crius v1 ) .

smurfy, 2019-09-10:
Board type is mainly for sensors.. crius1/2 both use MPU6000_I2C so it should be fine.

But the project is quite old, cant really help regarding mavlink problem, i just dont remember all the quirks anymore.

Osama-321, 2019-09-10:
Thank you ... but do you think if I connect the required sensors (compass , paro ) , the problem will be solved ?